yichen928/SparseFusion #2: About the environment dependencies

qzhang852, 2023-05-25:
Dear author, 
Thank you for your excellent work, I wonder if it is possible for you to update the loacl environment dependencies list like pytorch and numpy as well. I have tried to create the dockerfile for your work, and it seems that there are some conflicts.

yichen928, 2023-05-25:
I have updated the environment in README: "We test our code on an environment with CUDA 11.5, python 3.7, PyTorch 1.7.1, TorchVision 0.8.2, NumPy 1.20.0, and numba 0.48.0."

This numpy version may lead to conflict, but it does affect the use of code.

qzhang852, 2023-05-25:
Thank you for your swift reply. I have downgraded the pandas, and it works now. It could be because I installed mmdetection3d directly from your repository.

closed 2023-05-25 by qzhang852