yichen928/SparseFusion #3: questions about cuda version

gubei1998, 2023-06-05:
I find that cuda 11.5 does not support pytorch 1.7.1，so I want to know how did you installed pytorch 1.7.1 with cuda 11.5?Thank you very much,and the errors reported are as follows.
 

$ conda install pytorch==1.7.1 torchvision==0.8.2 torchaudio==0.7.2 cudatoolkit=11.5 -c pytorch
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - cudatoolkit=11.5

Current channels:

  - https://conda.anaconda.org/pytorch/linux-64
  - https://conda.anaconda.org/pytorch/noarch
  - https://repo.anaconda.com/pkgs/main/linux-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/linux-64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

yichen928, 2023-06-05:
Thanks for you interest.

You can use pip to install pytorch and torchvision. For example, "pip install torch==1.7.1+cu110 torchvision==0.8.2+cu110 torchaudio==0.7.2 -f https://download.pytorch.org/whl/torch_stable.html". 

It can be compatible with CUDA 11.X (e.g. 11.3, 11.5, 11.7).

gubei1998, 2023-06-05:
ok,I will try it,thank you very much!

yichen928, 2023-06-07:
Can you share the complete error message?

yichen928, 2023-06-08:
I would recommend you to install mmdet3d from their source code instead of directly using pip install.

yichen928, 2023-06-11:
Since there is no further reply, I temporally close this issue. Feel free to re-open it, if you have further questions.

closed 2023-06-11 by yichen928

gubei1998, 2023-06-13:
> I would recommend you to install mmdet3d from their source code instead of directly using pip install.

yes,I truely install mmdet3d from their source code, but it still reports this error.

yichen928, 2023-06-13:
It would be helpful if you can give the entire error message, and version of your cuda, pytorch, mmcv, mmdet, mmdet3d. Your above information does not include many details.