yichen928/SparseFusion #5: mmdet3d安装失败

xinpuliu, 2023-07-04:
你好，我再安装好了mmcv和mmdet之后，准备git了mmdet3d的源码进行安装的时候，报了下面的错误，您知道是为什么吗？
[image]
其他环境是：

xinpuliu, 2023-07-04:
其他环境是：
cuda 11.0
torch 1.7.0
numpy和numba和mmcv和mmdet都和您给出的一致。

yichen928, 2023-07-04:
你好！我没有遇到过这个问题，上网查询感觉可能是因为库版本不兼容。我注意到你好像没有使用conda，建议你可以安装miniconda (https://docs.conda.io/en/latest/miniconda.html)，之后创建一个新的conda环境再进行安装，

xinpuliu, 2023-07-04:
你好，我这个有conda环境的，新建conda环境进行安装，也遇到过相同的问题，我想问一下，您装mmcv，mmdet，和mmdet3d都是用什么指令装的呀？

xinpuliu, 2023-07-04:
怎样是安装历史版本的mmlab其中的库的最佳方式呢？

yichen928, 2023-07-05:
对于mmcv （命令中cuda和torch版本根据你实际安装版本确定）：
`pip install mmcv-full==1.2.7 -f https://download.openmmlab.com/mmcv/dist/cu115/torch1.7.1/index.html`

对于mmdet：
`pip install mmdet==2.10.0`

对于mmdet3d，建议从源代码安装:
```
git clone https://github.com/open-mmlab/mmdetection3d.git
cd mmdetection3d
git checkout v0.10.0
pip install -e .
```
尽量避免在同一个cuda环境中多次安装不同版本的上述库。

xinpuliu, 2023-07-05:
你好  我按照这个指令成功装上了mmcv  mmdet和mmdet3d这三个包，但是在运行训练指令的时候，报了这个错，请问是什么原因呀？

[image]

xinpuliu, 2023-07-05:
[image]

yichen928, 2023-07-05:
你可以尝试一下
`pip uninstall mmdet3d`
之后删除掉现有的mmdetection3d文件夹，然后在我们SparseFusion的目录下运行
`pip install -e .`

yichen928, 2023-07-05:
> 对于mmcv （命令中cuda和torch版本根据你实际安装版本确定）： `pip install mmcv-full==1.2.7 -f https://download.openmmlab.com/mmcv/dist/cu115/torch1.7.1/index.html`
> 
> 对于mmdet： `pip install mmdet==2.10.0`
> 
> 对于mmdet3d，建议从源代码安装:
> 
> ```
> git clone https://github.com/open-mmlab/mmdetection3d.git
> cd mmdetection3d
> git checkout v0.10.0
> pip install -e .
> ```
> 
> 尽量避免在同一个conda环境中多次安装不同版本的上述库。

这里我写错了，mmdet3d应为v0.11.0，但这应该不是这个问题的原因。

xinpuliu, 2023-07-05:
谢谢您的耐心解答，按照在SparseFusion的目录下编译，已经可以运行代码了。谢谢

另外我想问一下训练时间的问题，你们的工作用的是4张a6000，那训完swinT那6个epoch大概需要几天呢？

yichen928, 2023-07-05:
ResNet50和SwinT都大约需要两天

closed 2023-07-05 by yichen928

xioatian1, 2023-07-10:
你好，我按照你们所讨论安装环境的问题去安装我的环境
我的cuda版本是11.0，torch=1.7.0
使用指令的过程如下：
`conda install pytorch==1.7.0 torchvision==0.8.0 torchaudio==0.7.0 cudatoolkit=11.0 -c pytorch`
`pip insatll Numpy==1.20.0`
`pip insatll numba==0.48.0`

`pip install mmcv-full==1.2.7 -f https://download.openmmlab.com/mmcv/dist/cu110/torch1.7.0/index.html`

`pip install mmdet==2.10.0`

`git clone https://github.com/open-mmlab/mmdetection3d.git`
`cd mmdetection3d`
`git checkout v0.11.0`
`pip install -e .`
但是执行到安装mmdetection3d这一步的时候总显示显示安装错误，您能帮我看看是哪个步骤有问题吗？
[image]

yichen928, 2023-07-11:
你这个错误信息貌似没有截图完整，需要看一下前面的部分才能判断原因。此外，你也可以尝试直接在我们的仓库里运行pip install -e .，而无需额外安装mmdet3d。

xioatian1, 2023-07-12:
> 你这个错误信息貌似没有截图完整，需要看一下前面的部分才能判断原因。此外，你也可以尝试直接在我们的仓库里运行pip install -e .，而无需额外安装mmdet3d。

        非常感谢您的回复！我是直接在你们的仓库里面运行pip install -e .，现在已经成功安装了环境。但是当我开始处理数据集的时候，出现了spconv模块未安装的问题，因为我的cuda版本是11.0，好像不能用spconv 2.x ,只能用spconv 1.x版本的，您是用什么版本的spconv的呢？期待您的回复。
[image]

yichen928, 2023-07-12:
你可以安装pip install spconv-cu113试试（应该是可以的），如果确实不行可以参考https://github.com/XuyangBai/TransFusion/issues/10#issuecomment-1118361133，反向操作替换为spconv1.x，相应有一些initial weight参数名也需要修改。

xioatian1, 2023-07-14:
> 你可以安装pip install spconv-cu113试试（应该是可以的），如果确实不行可以参考[https://github.com/XuyangBai/TransFusion/issues/10#issuecomment-1118361133。](https://github.com/XuyangBai/TransFusion/issues/10#issuecomment-1118361133%E3%80%82) 反向操作替换为spconv1.x，相应有一些initial weight参数名也需要修改。

感谢您的回复！我将按照您的方法去修改

BoomSky0416, 2023-07-24:
@xioatian1 请问一下你的gcc版本是多少

yichen928, 2023-07-24:
> @xioatian1 请问一下你的gcc版本是多少

我使用的是GCC 7.3.0

BoomSky0416, 2023-07-25:
> 你好，我再安装好了mmcv和mmdet之后，准备git了mmdet3d的源码进行安装的时候，报了下面的错误，您知道是为什么吗？ [image] 其他环境是：

@xinpuliu 你好，请问你这个问题是怎么解决的，我现在安装mmdet3d遇到了相同的问题

BoomSky0416, 2023-07-25:
@yichen928 请问你的环境可以上传dockerhub吗？

xioatian1, 2023-07-25:
> > 你好，我再安装好了mmcv和mmdet之后，准备git了mmdet3d的源码进行安装的时候，报了下面的错误，您知道是为什么吗？ [image] 其他环境是：
> 
> @xinpuliu 你好，请问你这个问题是怎么解决的，我现在安装mmdet3d遇到了相同的问题

我认为在安装mmdet3d之前你可以试试安装ninja，或者你可以直接用作者提供的仓库代码来安装环境

BoomSky0416, 2023-07-25:
> > > 你好，我再安装好了mmcv和mmdet之后，准备git了mmdet3d的源码进行安装的时候，报了下面的错误，您知道是为什么吗？ [image] 其他环境是：
> > 
> > 
> > @xinpuliu 你好，请问你这个问题是怎么解决的，我现在安装mmdet3d遇到了相同的问题
> 
> 我认为在安装mmdet3d之前你可以试试安装ninja，或者你可以直接用作者提供的仓库代码来安装环境

好的，非常感谢，请问ninja是哪个版本呢？

xioatian1, 2023-07-25:
> 

我是默认最新版本，这个应该没有影响

xioatian1, 2023-07-25:
> > > > 你好，我再安装好了mmcv和mmdet之后，准备git了mmdet3d的源码进行安装的时候，报了下面的错误，您知道是为什么吗？ [image] 其他环境是：
> > > 
> > > 
> > > @xinpuliu 你好，请问你这个问题是怎么解决的，我现在安装mmdet3d遇到了相同的问题
> > 
> > 
> > 我认为在安装mmdet3d之前你可以试试安装ninja，或者你可以直接用作者提供的仓库代码来安装环境
> 
> 好的，非常感谢，请问ninja是哪个版本呢？

我是默认最新版本，这个应该没有影响

BoomSky0416, 2023-07-25:
我刚刚参考了这个https://github.com/open-mmlab/mmdetection3d/issues/362#issuecomment-1104959006，问题解决了

SISTMrL, 2023-07-30:
你好请问下cuda11版本该如何该安装呢，感觉各种版本兼容问题

SISTMrL, 2023-07-30:
> pip install mmcv-full==1.2.7 -f https://download.openmmlab.com/mmcv/dist/cu115/torch1.7.1/index.html

你好我本地cuda版本11.7，安装torch的指令是conda install pytorch==1.7.1 torchvision==0.8.2 torchaudio==0.7.2 cudatoolkit=11.0

我的mmcv安装pip install mmcv-full==1.2.7 -f https://download.openmmlab.com/mmcv/dist/cu117/torch1.7.1/index.html

但是报错了，请问下怎么解决，十分感谢，我配了好久都没搞定，(;´༎ຶД༎ຶ`) 

[image]

yichen928, 2023-07-30:
> 你好我本地cuda版本11.7，安装torch的指令是conda install pytorch==1.7.1 torchvision==0.8.2 torchaudio==0.7.2 cudatoolkit=11.0
>
> 我的mmcv安装pip install mmcv-full==1.2.7 -f [...]

你可以尝试一下使用pip安装pytorch，`pip install torch==1.7.1+cu110 torchvision==0.8.2+cu110 -f https://download.pytorch.org/whl/torch_stable.html`

sidiangongyuan, 2024-05-29:
你好， 我的cuda 版本是11.6， mmcv 和 mmdet 都可以正常安装，但是mmdet3D，怎么也装不上。不管是在你的项目下 ，还是 在mmdetection3D项目下编译，都报相同的问题

[image]

[image]

[image]

puttii, 2026-02-03:
> 你好， 我的cuda 版本是11.6， mmcv 和 mmdet 都可以正常安装，但是mmdet3D，怎么也装不上。不管是在你的项目下 ，还是 在mmdetection3D项目下编译，都报相同的问题
> 
> [image]
> 
> [image]
> 
> [image]

你好，解决了吗

sidiangongyuan, 2026-02-04:
时间太过久远了， 当时应该是解决了， 可以升级版本， 需要查看 mmcv / mmdet 的版本支持 哪些 版本的 mm3d。