yichen928/SparseFusion #8: AttributeError: 'DataContainer' object has no attribute 'shape'

BoomSky0416, 2023-07-25:
I encounter this error during inference. Here is my env.
[image]

yichen928, 2023-07-25:
Hi Thanks for your interest. I did not encounter this error. Can you share more details such as the error message or the concrete line causing this error?

BoomSky0416, 2023-07-26:
Thanks for your response. Here is the screenshot of the error message.
[image]

yichen928, 2023-07-26:
Can you check that 1) Are you using PyTorch 1.7 now? 2) Are your GPUs available by checking `torch.cuda.is_available()`?

BoomSky0416, 2023-07-26:
Oh, my Pytorch version is 1.9.0

yichen928, 2023-07-26:
Please consider re-installing your environment with PyTorch 1.7

closed 2023-07-27 by BoomSky0416