yichen928/SparseFusion #9: install he mmdet==2.10.0

SISTMrL, 2023-07-31:
你好，机器的cuda版本是11.7
安装torch的指令是：pip install torch==1.7.1+cu110 torchvision==0.8.2+cu110 -f https://download.pytorch.org/whl/torch_stable.html
安装mmcv-full的指令是：pip install mmcv-full==1.2.7 -f https://download.openmmlab.com/mmcv/dist/cu117/torch1.7.1/index.html
numpy==1.20.0
numba==0.48.0

但是执行pip install mmdet==2.10.0，出现fail to build mmpycocotools，请问下该如何解决呢？(;´༎ຶД༎ຶ`) 谢谢！

SISTMrL, 2023-07-31:
我下载了mmdetection 2.10.0的源码进行编译，还是出现fail to build mmpycocotools(;´༎ຶД༎ຶ`)

SISTMrL, 2023-07-31:
最新的mmdet不需要mmpycocotools，但是装上后，mmcv的版本就不能用了

yichen928, 2023-08-01:
I think you may try to install mmpycocotools manually by running the following code:
`pip uninstall pycocotools`
`pip uninstall mmpycocotools`
`pip install mmpycocotools`

SISTMrL, 2023-08-01:
> I think you may try to install mmpycocotools manually by running the following code: `pip uninstall pycocotools` `pip uninstall mmpycocotools` `pip install mmpycocotools`

hi, it doesn't work, do you have any ideas?

yichen928, 2023-08-01:
https://github.com/cocodataset/cocoapi/issues/172
Maybe this issue can help.

SISTMrL, 2023-08-01:
> [cocodataset/cocoapi#172](https://github.com/cocodataset/cocoapi/issues/172) Maybe this issue can help.

thanks, i will have a try

SISTMrL, 2023-08-01:
> [cocodataset/cocoapi#172](https://github.com/cocodataset/cocoapi/issues/172) Maybe this issue can help.

thank you, i found the correct way to install mmdet in the link you provided. Just use conda to install cython, not pip

closed 2023-08-01 by SISTMrL