yichen928/SparseFusion #10: About Fig. 6.

Raiden-cn, 2023-08-02:
What a AWESOME work!
BTW, Fig 6 is really brilliant! 
Could give a hint or some adivecs about how to draw it?
Code maybe is best. : )
Looking for your reply!

yichen928, 2023-08-03:
Hi, thanks for your interest! We give a detailed explanation of this figure in Appendix A3. The code to draw this paper is attached as follows. The detected results of each branch and attention strength are saved beforehand.
[visualize_attn.txt](https://github.com/yichen928/SparseFusion/files/12247568/visualize_attn.txt)

dodeokwangyooo, 2025-02-22:
> 您好，感谢您的关注！我们在附录 A3 中对这个数字进行了详细说明。绘制本文的代码附后如下。每个分支的检测结果和注意力强度都会事先保存。[visualize_attn.txt](https://github.com/yichen928/SparseFusion/files/12247568/visualize_attn.txt)

您好！这个图真的太棒了！但是看您的代码还是有些疑惑？请问输入的pkl文件是怎么生成的呢？

yichen928, 2025-02-22:
> > 您好，感谢您的关注！我们在附录 A3 中对这个数字进行了详细说明。绘制本文的代码附后如下。每个分支的检测结果和注意力强度都会事先保存。[visualize_attn.txt](https://github.com/yichen928/SparseFusion/files/12247568/visualize_attn.txt)
> 
> 您好！这个图真的太棒了！但是看您的代码还是有些疑惑？请问输入的pkl文件是怎么生成的呢？

我记得mmdetection3d的默认inference代码里面有一个保存结果的选项，应该就是这里保存下来的pkl文件

dodeokwangyooo, 2025-02-23:
> > > 您好，感谢您的关注！我们在附录 A3 中对这个数字进行了详细说明。绘制本文的代码附后如下。每个分支的检测结果和注意力强度都会事先保存。[visualize_attn.txt](https://github.com/yichen928/SparseFusion/files/12247568/visualize_attn.txt)
> > 
> > 
> > 您好！这个图真的太棒了！但是看您的代码还是有些疑惑？请问输入的pkl文件是怎么生成的呢？
> 
> 我记得mmdetection3d的默认inference代码里面有一个保存结果的选项，应该就是这里保存下来的pkl文件

太感谢您的回复了！还有个问题就是文件中的attn_weights.pth是对应哪部分的权重呢？

yichen928, 2025-02-25:
> 太感谢您的回复了！还有个问题就是文件中的attn_weights.pth是对应哪部分的权重呢？

对应最后一层多模态的self-attention权重