yichen928/SparseFusion #11: GPU memory consumption is high

minrui-hust, 2023-08-02:
Can not finish training with 3090(24GB) even set batch_size to 1, any advise?

yichen928, 2023-08-03:
Hi! Thanks for your interest. 
You can simply reduce the GPU memory without any damage to performance by editing the `generate_heatmap_deform` function in `mmdet3d/models/dense_heads/sparsefusion_head_deform.py`. The following part can run on each view separately in a for-loop.

https://github.com/yichen928/SparseFusion/blob/89a7c5a9e2a1bd1bf63631819af8a78c69df744c/mmdet3d/models/dense_heads/sparsefusion_head_deform.py#L705

In this case, you do not need to pad the input queries as in https://github.com/yichen928/SparseFusion/blob/89a7c5a9e2a1bd1bf63631819af8a78c69df744c/mmdet3d/models/dense_heads/sparsefusion_head_deform.py#L671

This would not harm the performance but may very slightly slow down the inference, so you can also only modify this during training. We will also update a version with this change soon.

minrui-hust, 2023-08-04:
Thanks for your advise, I'll try it. 
BTW ,would you provide your training logs for reference, currently, only ckpt and config is available

yichen928, 2023-08-04:
This is a log file of our training (ResNet50 backbone) [20230717_235818.log](https://github.com/yichen928/SparseFusion/files/12257281/20230717_235818.log). Please notice that this is a different training with our provided checkpoint, so the final performance is slightly different.

closed 2023-08-04 by yichen928

BoomSky0416, 2023-08-08:
@yichen928 Hello, I have two question about this change for gpu memory reduce. Is it necessary to set up multiple cross_heatmap_decoders for each view? And how to only modify this during training, what's the difference between training and inference?

SISTMrL, 2023-08-15:
> 
yichen，你好，很抱歉看了你的回答后，还是不清楚该怎么修改代码，是在做cross attn的时候，执行红框部分的代码吗
[image]

yichen928, 2023-08-22:
> > 
> 
> yichen，你好，很抱歉看了你的回答后，还是不清楚该怎么修改代码，是在做cross attn的时候，执行红框部分的代码吗？ 另外改完后，可以直接resume吗，模型参数会不会改，我训到第三个epoch，显存爆了 [image]

I have updated a new version of the code with much better GPU efficiency. It would be helpful and more identifiable to re-open the issue when you have any following questions.