yichen928/SparseFusion #14: cuda error

SISTMrL, 2023-08-04:
你好这个是我遇到的错误RuntimeError: CUDA error: no kernel image is available for execution on the device

gpu：a40, cuda 11.7
torch 1.7.1+cu11.0, torchvision:0.8.2+cu110
mmdet==2.10.0
mmcv==1.2.7
NumPy 1.20.0
numba 0.48.0.
spconv-cu117 2.3.6
跟你不同的就是所用的gpu， cuda版本，以及spconv
请问下你有哪些建议吗，十分感谢！！！！

yichen928, 2023-08-06:
这个问题看上去是cuda driver或者cuda toolkit安装的问题，你可以试试`torch.cuda.is_available()`检查一下

SISTMrL, 2023-08-07:
> 这个问题看上去是cuda driver或者cuda toolkit安装的问题，你可以试试`torch.cuda.is_available()`检查一下

查过了，是true

yichen928, 2023-08-09:
我觉得你可以尝试一下安装一个cuda 11.1的[toolkit](https://developer.nvidia.com/cuda-11.1.0-download-archive)（不需要重新安装driver），然后再尝试安装一下

SISTMrL, 2023-08-10:
> 我觉得你可以尝试一下安装一个cuda 11.1的[toolkit](https://developer.nvidia.com/cuda-11.1.0-download-archive)（不需要重新安装driver），然后再尝试安装一下

好的谢谢，我试一试，十分感谢！！！！！！

SISTMrL, 2023-08-10:
你好，方便问一下freeze lidar后训练整个模型，6个epoch大概需要多久

SISTMrL, 2023-08-11:
你好我在网上没有找到这篇文章的补充材料，方便能给补充材料吗，谢谢

closed 2023-08-11 by SISTMrL