yichen928/SparseFusion #15: generate depth

SISTMrL, 2023-08-11:
你好，我看了你从点云生成深度的代码，对于你生成valid_mask的部分感到困惑，为啥使用torch.clamp把小于0的depth变为0，我的理解是depth小于0应该是投影到了图像的后面，应该去除torch.clamp这行代码，然后valid mask的代码应该是：

valid_mask = (points_2d_view[:, 0] > 0) & (points_2d_view[:, 0] < w_shape-1) & \
(points_2d_view[:, 1] > 0) & (points_2d_view[:, 1] < h_shape-1) & \
(depth >= 1e-4)

大佬能解释下我的疑惑吗，十分感谢！！

[image]

SISTMrL, 2023-08-11:
你好我还有一个问题，为啥你这里的深度是2 x h x w的维度，不应该是1吗，你在[1, cy, cx]设置为1，然后在下图中进行相乘，不是很理解这样做的原因是啥？
另外这里深度的均值方差参数的设置有什么讲究吗

[image]

yichen928, 2023-08-12:
> 你好，我看了你从点云生成深度的代码，对于你生成valid_mask的部分感到困惑，为啥使用torch.clamp把小于0的depth变为0，我的理解是depth小于0应该是投影到了图像的后面，应该去除torch.clamp这行代码，然后valid mask的代码应该是：
>
> valid_mask = (points_2d_view[:, 0] > 0) & (points_2d_view[:, [...]

因为负的深度全部变成1e-4了，所以除完之后xy坐标基本都会超过上界，不过你说的做法确实是更好的，可以避免极少数xy很小的特殊情况。

closed 2023-08-12 by SISTMrL

yichen928, 2023-08-12:
> 你好我还有一个问题，为啥你这里的深度是2 x h x w的维度，不应该是1吗，你在[1, cy, cx]设置为1，然后在下图中进行相乘，不是很理解这样做的原因是啥？ 另外这里深度的均值方差参数的设置有什么讲究吗
> 
> [image]

这里深度是稀疏的雷达点云的投影，所以不能保证每个位置都有ground-truth深度，所以我用了二维特征，第一维是归一化后的深度，第二维是0/1，表示这个位置是不是有ground-truth深度。均值和方差应该是我在整个训练集上算的。