yichen928/SparseFusion #17: load数据新增的5个信息

xinpuliu, 2023-08-23:
熠辰你好，非常感谢你开源这么好的工作的代码，在调试的时候，我有些许疑问想询问下您。主要关于刚开始load数据的时候，新增的5个信息分别是什么含义呀？我看你专门自己写了pipline的代码。

xinpuliu, 2023-08-23:
这五个变量分别是：
gt_pts_centers_view
gt_img_centers_view
gt_bboxes_cam_view
gt_bboxes_lidar_view
gt_visible_3d
谢谢啦
[image]
[image]

yichen928, 2023-08-23:
我们的代码里load了两个坐标系下的ground-truth box，一个是lidar坐标系也就是原来的（M个物体），另一个是六个相机坐标系（N个物体），这两套GT并非一一对应，因为一个3D bbox可能出现在零个一个或者两个相机视野内。

这里gt_visible_3d对应LiDAR坐标系下的GT（M个物体）：
gt_visible_3d：每个3D bbox能否出现在至少一个相机视野内；

其他都对应Camera坐标系下的GT（N个物体）：
gt_bboxes: 物体对应2D bounding box
gt_labels: 二维，第一维是物体类别，第二维是拍摄到物体的相机编号
gt_img_centers_view：物体3D中心投影到2D的坐标，第三维是深度
gt_pts_centers_view：物体3D中心坐标（LiDAR坐标系）
gt_bboxes_cam_view：物体在相应相机坐标系下3D bbox
gt_bboxes_lidar_view: 物体在LiDAR坐标系下3D bbox（可能有重复，因为是与相机坐标系下物体一一对应）

xinpuliu, 2023-08-23:
谢谢你的解答，非常清晰。

closed 2023-08-23 by xinpuliu