yichen928/SparseFusion #21: About your combine_view_info

Stay-Naive, 2023-09-30:
Hi! Thanks for your great work and open source code.

I just have a question about your view infomation combination.  In your code base, you projected 3D bounding boxes to image plane to get 2D bbox annotations. It looks like you've taken all the projection results that are completely contained on the 2D image plane as 2D annotations. I want to know what about those partially appear on the plane of the 2D image? Do you just define those as invalid?

I want to know this because I came across an error when I re-use your code in NuScenes dataset. I find a sample in which I will get only 1 3D annotation, however this annotation will be treated as invalid because it has no complete 2D projection. It will raise error in `SparseFusionHead2D_Deform.get_targets_single_view` and line ` max_overlaps=torch.cat([res.max_overlaps for res in assign_result_list]),` since here the only annotation will be filtered.

`expected Tensor as element 0 in argument 0, but got NoneType`

The sample's token is 'd3f0e2be2cfc4a9ebf6e28a08d3ebab2', and the only one 3D annotation's token is '8ff497264f794fa98f9d762abbb3760c'.  You may use `nusc.get('sample', sample_token)` and `annotation_metadata =  nusc.get('sample_annotation', annotation_token)` to get them, and `nusc.render_annotation(annotation_token, extra_info=True)`  to visualize it.

I wonder if you have met this problem, since I have not modified your data loading pipeline and basically re-use your dense head code base. And could you give me some advice about how to address this problem?

Stay-Naive, 2023-09-30:
[image]
here is the annotation's visualization.

yichen928, 2023-10-03:
Thanks very much for your interest! 
1. For objects projected into multiple views, each view would have its own corresponding bounding box based on the projected object. All of these bounding boxes would be valid.

2. I have not encountered this problem. Can you check whether you Hungarian Assigner is same as ours? Especially for the special case when num_gts=0.

Stay-Naive, 2023-10-04:
My bad. I checked my code and I found some parts of my code were inherited from TransFusion which were a little different from yours.

Thanks for your useful advise! I will close this issue.

closed 2023-10-04 by Stay-Naive