yichen928/SparseFusion #23: create_data.py

lmh0501, 2023-11-02:
你好，想问一下，运行python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes时出现这个问题该怎么解决呢？
[image]

yichen928, 2023-11-02:
感谢提问！我没有遇到这个问题，你可以检查一下numpy和numba的版本是否正确，我们使用的是numpy 1.20.0, numba 0.48.0

lmh0501, 2023-11-02:
[image]
是一样的。我现在是用nuscenes中的mini数据集，会跟这个有关系嘛？

yichen928, 2023-11-02:
你可以尝试一下把numpy版本降低到1.19.5

lmh0501, 2023-11-02:
好像不大行，我试了numpy和numba的不同版本，不同情况的报错，只有你给的这个版本能运行到上面这一步然后也是会报错

lmh0501, 2023-11-02:
> 好像不大行，我试了numpy和numba的不同版本，不同情况的报错，只有你给的这个版本能运行到上面这一步，然后也是会报错

我用的单张3090显卡，定位到报错的那行，然后把最后的dtype=np.bool改成dtype=np.int64，代码就能跑起来了，但是构建GT Database这步运行得特别慢，我看了一下大概需要七个小时，这还是mini数据集，是不是不大对啊？

yichen928, 2023-11-04:
我们在完整数据集上确实是需要六七个小时，但是mini数据集我不太确定，你们可以确认一下是不是确实只构建了mini数据集，还是其实是在构建完整数据集。

closed 2023-11-06 by lmh0501

xtk8532704, 2024-06-01:
同样的问题。
```
Create GT Database of NuScenesDataset
[                                                  ] 0/28130, elapsed: 0s, ETA:Traceback (most recent call last):
  File "tools/create_data.py", line 225, in <module>
    max_sweeps=args.max_sweeps)
  File "tools/create_data.py", line 67, in nuscenes_data_prep
    f'{out_dir}/{info_prefix}_infos_train.pkl')
  File "/home/palmdong/mmfusion/SparseFusion/tools/data_converter/create_gt_database.py", line 254, in create_groundtruth_database
    point_indices = box_np_ops.points_in_rbbox(points, gt_boxes_3d)
  File "/home/palmdong/mmfusion/SparseFusion/mmdet3d/core/bbox/box_np_ops.py", line 433, in points_in_rbbox
    indices = points_in_convex_polygon_3d_jit(points[:, :3], surfaces)
  File "/home/palmdong/mmfusion/SparseFusion/mmdet3d/core/bbox/box_np_ops.py", line 763, in points_in_convex_polygon_3d_jit
    normal_vec, d, num_surfaces)
TypeError: expected dtype object, got 'numpy.dtype[bool_]'
```

我的版本：
```
(sparsefusion) palmdong@473370ef5807:~/mmfusion/SparseFusion$ pip list 
Package                       Version     Editable project location
----------------------------- ----------- ------------------------------------
absl-py                       2.1.0
addict                        2.4.0
anyio                         3.7.1
argon2-cffi                   23.1.0
argon2-cffi-bindings          21.2.0
attrs                         23.2.0
backcall                      0.2.0
beautifulsoup4                4.12.3
black                         23.3.0
bleach                        6.0.0
cachetools                    5.3.3
ccimport                      0.4.2
certifi                       2022.12.7
cffi                          1.15.1
charset-normalizer            3.3.2
click                         8.1.7
comm                          0.1.4
cumm-cu117                    0.4.11
cycler                        0.11.0
Cython                        0.29.33
debugpy                       1.7.0
decorator                     5.1.1
defusedxml                    0.7.1
descartes                     1.1.0
entrypoints                   0.4
exceptiongroup                1.2.1
fastjsonschema                2.19.1
fire                          0.6.0
flake8                        3.9.2
fonttools                     4.38.0
google-auth                   2.29.0
google-auth-oauthlib          0.4.6
grpcio                        1.62.2
idna                          3.7
imageio                       2.31.2
importlib-metadata            6.7.0
importlib-resources           5.12.0
iniconfig                     2.0.0
ipykernel                     6.16.2
ipython                       7.34.0
ipython-genutils              0.2.0
ipywidgets                    8.1.3
jedi                          0.19.1
Jinja2                        3.1.4
joblib                        1.3.2
jsonschema                    4.17.3
jupyter                       1.0.0
jupyter_client                7.4.9
[... 94 lines cut ...]
terminaltables                3.1.10
threadpoolctl                 3.1.0
tifffile                      2021.11.2
tinycss2                      1.2.1
tomli                         2.0.1
torch                         1.7.1+cu110
torchaudio                    0.7.2
torchvision                   0.8.2+cu110
tornado                       6.2
tqdm                          4.66.4
traitlets                     5.9.0
trimesh                       2.35.39
typed-ast                     1.5.5
typing_extensions             4.7.1
urllib3                       2.0.7
wcwidth                       0.2.13
webencodings                  0.5.1
websocket-client              1.6.1
Werkzeug                      2.2.3
wheel                         0.38.4
widgetsnbextension            4.0.11
yapf                          0.40.2
zipp                          3.15.0

```

xtk8532704, 2024-06-01:
解决了,降版本+需要重新编译mmpycocotools:

pip uninstall numpy; pip install numpy==1.20.0 #需要在mmpycocotools前面装。
pip uninstall mmpycocotools pycocotools; pip install mmpycocotools==12.0.3 --no-cache-dir

想问一嘴为何用numpy==1.20.0,有什么讲究么？

dodeokwangyooo, 2025-01-12:
> 解决了,降版本+需要重新编译mmpycocotools:
> 
> pip uninstall numpy; pip install numpy==1.20.0 #需要在mmpycocotools前面装。 pip uninstall mmpycocotools pycocotools; pip install mmpycocotools==12.0.3 --no-cache-dir
> 
> 想问一嘴为何用numpy==1.20.0,有什么讲究么？

我运行pip uninstall numpy; pip install numpy==1.20.0 然后再 pip uninstall mmpycocotools pycocotools; pip install mmpycocotools==12.0.3 --no-cache-dir还是会报错离谱 就是bool那个错