yichen928/SparseFusion #24: The problem with FPS during inferencing

liyih, 2024-01-02:
Hi, yichen,
Thanks for your awesome work. I use A100 to inference your Resnet-50 based model, and only get FPS=2.1. It's strange, as in your table 4, you got FPS=5.6 with A6000. I use tools/analysis_tools/benchmark.py for inference. I want to know how do you calculate the FPS in your model.
Best

liyih, 2024-01-03:
I guess you don't consider voxelization time just as bevfusion(MIT) do. I remove the voxelization time and got fps=6.4 in A100, that looks good.

yichen928, 2024-01-03:
> I guess you don't consider voxelization time just as bevfusion(MIT) do. I remove the voxelization time and got fps=6.4 in A100, that looks good.

That is correct! Thanks for your interest.

closed 2024-01-03 by yichen928