yichen928/SparseFusion #31: The results of Transfusion

Wang-pengfei, 2024-03-16:
Hi Yichen, I really admire your impressive work! I noticed a slight discrepancy in the results for "Transfusion" in Table 2 compared to the original text. Could it be because of using a different image backbone? Did you reproduce these results yourself?

yichen928, 2024-03-16:
Our TransFusion results come from Tab. 1 of DeepInteraction paper (https://arxiv.org/pdf/2208.11112.pdf). I think their validation set results come from Tab. 11 of TransFusion paper (https://arxiv.org/pdf/2203.11496.pdf). Their test set results are same as nuScenes leaderboard. 
[image]

Wang-pengfei, 2024-03-16:
Thank you for your nice response！

There might be an error in the backbone mentioned in the Tab. 1 of DeepInteraction? Instead of ResNet50, Transfusion actually uses DLA34. Or did I miss something? 

I have one more question regarding Table 3 in your paper.  Did you modify the query initialization of TransFusion-L to obtain the L+ST results? Also, did you continue training TransFusion-LC after modifying the query initialization? I'm curious to know how the results about it.

yichen928, 2024-03-16:
Based on the official source code of TransFusion, they use ResNet50 as the image backbone and achieve the same performance as their paper. I think this may be a typo in their paper.

For the L+ST results in Tab. 3, we directly train our full SparseFusion model, but extract the detection results in the LiDAR branch instead of the fusion stage.

Wang-pengfei, 2024-03-17:
Thank you for your response, I got it.

closed 2024-03-17 by Wang-pengfei