yichen928/SparseFusion #32: Download nuScenes dataset

Andhow0603, 2024-04-16:
Hi, Yichen, thank you for your great work!

I faced some question about downloading nuSecens dataset.
Since the official website from nuScenes covers many datasets.

Q1: In fugure1, I didn't know which dataset need to download in full dataset.
Q2: in figure2, Does the Metadata need to download in Test dataset? Will Metadata be used in Sparsefusion?

Looking forward to your reply!!
thanks, Howard

[image]
[image]

yichen928, 2024-04-16:
Thanks for your interest!

To reproduce our results on the validation set, you should download and unzip all the zip files for Trainval, including the meta data.

If you want to submit your results to nuScenes leaderboard, you should also download the Test dataset. I believe this would only be useful in the very late stage of your research project.

Andhow0603, 2024-04-17:
Thank you for your reply.
However, I found that nuScenes covers US and Asia. Do I need to download both regions? Or only one of them?

yichen928, 2024-04-17:
They are same data from different sources, so you only need to download one of them. For example, if you are in Asia, the Asia source may be faster to download.

Andhow0603, 2024-04-18:
Thank you for your help.
Sorry, I want to ask you some questions.

1. In paper had mentioned that "The multi-view sparse depth map is processed by the shared encoder to obtain the depth feature”，Obtaining the sparse feature from the sparse depth map is through sparse CNN or general CNN.?

2. When I have opened the terminal in the tool folder, enter this command "bash dist_test.sh ../configs/sparsefusion_nusc_voxel_LC_r50.py /home/lab528/Desktop/howard/SparseFusion/checkpoints/sparsefusion_voxel0075_R50_initial.pth 4 --eval= bbox"
but i got some errors about this:
[image]
[image]
I check the version in "pip list" 'mmcv-full' version is 1.2.7 and 'mmdet' version is 2.10.0, 'mmdet3d' version is 0.11.0.
I also check the "mim list", it just shows that mmcv-full 1.2.7 and mmengine 0.10.3, didn't show the mmdet in here.
I'm wondering if this error has something to do with not installing mmdet into mim.

yichen928, 2024-04-24:
1. It is general CNN.
2. I think this is not related to our project, also not related to mmcv, mmdet, mmdet3d, and so on. You may double check the compatibility of your CUDA, NCCL, and PyTorch.

Andhow0603, 2024-04-26:
Thanks for your information, I will try.
Bless you have a nice day.

closed 2024-04-27 by yichen928