yichen928/SparseFusion #37: import DCN failed

ZhangYK618, 2024-07-05:
When I try to train with "bash tools/dist_train.sh configs/sparsefusion_nusc_voxel_LC_r50.py 4 --work-dir work_dirs/sparsefusion_nusc_voxel_LC_r50" I find that I get the same problem. LC_r50" training, I found that the error "import DCN failed" occurs, so I went to look for 13 lines of code in mmdet3d/models/backbones/DLA.py file "from dcn_v2 import DCN" error, I go to the network query and did not find a suitable solution, if you can see hope can help for the
[image]
[image]
[image]

ZhangYK618, 2024-07-05:
[image]
[image]

yichen928, 2024-07-12:
You can just ignore this failure message. This should not affect the code running.

closed 2024-07-15 by yichen928