yichen928/SparseFusion #39: Hi,

loserhou, 2024-07-22:
Hi, 
Thanks for your interest. Both checkpoints are merged by the pretrained weights of LiDAR and camera parts separately.

For both checkpoints, the LiDAR part is a pretrained LiDAR-only detector TransFusion-L, provided by the authors of DeepInteraction (https://drive.google.com/file/d/1IaLMcRu4SYTqcD6K1HF5UjfnRICB_IQM/view?usp=sharing). The backbone, FPN, and head are all incorporated into our initial weights.

For "sparsefusion_voxel0075_R50_initial", the ResNet-50 backbone and FPN is from Mask-RCNN pretrained on nuImage (https://download.openmmlab.com/mmdetection3d/v0.1.0_models/nuimages_semseg/mask_rcnn_r50_fpn_coco-2x_1x_nuim/mask_rcnn_r50_fpn_coco-2x_1x_nuim_20201008_195238-b1742a60.pth).  For "sparsefusion_voxel0075_SwinT_initial", the Swin-T backbone and FPN are further finetuned by us on nuImages from the COCO pretrained weights (https://download.openmmlab.com/mmdetection/v2.0/swin/mask_rcnn_swin-t-p4-w7_fpn_1x_coco/mask_rcnn_swin-t-p4-w7_fpn_1x_coco_20210902_120937-9d6b7cfa.pth).

You may need to change some parameter names when merging the LiDAR and camera parts to get our initial weights.

_Originally posted by @yichen928 in https://github.com/yichen928/SparseFusion/issues/4#issuecomment-1599259889_

loserhou, 2024-07-22:
您好，请问可以提供lidar分支pth文件修改参数名称以及两个pth文件怎么进行合并的具体过程吗？刚刚入门，不太理解，希望得到您的解答！

yichen928, 2024-07-23:
pth文件load之后其实是一个dict，合并的时候将两个dict合并成一个，并且存储为一个新的pth就可以。参数名称也就是字典的key，openmmlab不同代码库有些许的不同，例如ResNet的权重中参数名中的backbone在mmdet3d中要替换为img_backbone，具体可以我们提供的合并后的与合并前的权重的key都打印出来进行对比。

loserhou, 2024-07-24:
感谢解答！祝您工作顺利！

| |
侯业飞
***@***.***
|
---- Replied Message ----
| From | Yichen ***@***.***> |
| Date | 7/24/2024 03:46 |
| To | ***@***.***> |
| Cc | ***@***.***>,
***@***.***> |
| Subject | Re: [yichen928/SparseFusion] Hi, (Issue #39) |

pth文件load之后其实是一个dict，合并的时候将两个dict合并成一个，并且存储为一个新的pth就可以。参数名称也就是字典的key，openmmlab不同代码库有些许的不同，例如ResNet的权重中参数名中的backbone在mmdet3d中要替换为img_backbone，具体可以我们提供的合并后的与合并前的权重的key都打印出来进行对比。

closed 2024-07-24 by yichen928