yichen928/SparseFusion #41: The camera branch loss is high and iou is low

wayyeah, 2024-08-12:
Thank you for your great work. When I was training SparseFusion on my own data set, I found that the loss of the camera branch was very large and difficult to decrease. At the same time, the 2D iou was particularly low. What do you think the problem might be?
[image]

yichen928, 2024-08-19:
Based on my experience, you may double check the projection between the 3D coordinates and camera coordinates for  both the GT bboxes and internal network model.

wayyeah, 2024-08-19:
> Based on my experience, you may double check the projection between the 3D coordinates and camera coordinates for both the GT bboxes and internal network model.
当我训练单lidar分支的时候，我遇到了heatmaploss震荡的问题，我排查后发现是梯度爆炸导致网络参数出现了nan，你认为可能会是什么原因造成的？
[image]

yichen928, 2024-08-19:
这个我也没有遇到过，你可以试着降低learning rate以及clip grad norm

wayyeah, 2024-08-19:
> 这个我也没有遇到过，你可以试着降低learning rate以及clip grad norm
好滴，谢谢

closed 2024-08-19 by wayyeah