yichen928/SparseFusion #43: 关于mmdet3d的安装

Ge-huihui, 2024-11-12:
我的cuda版本是11.5  pytorch=1.7.1（使用指令   安装的，因pytorch官网上没有11.5版本对应的pytorch）  numpy=1.20.0   numba==0.48.0    mmdet==2.10.0  mmcv==1.2.7     都是按照作者您官方要求装的，但是到最后安装mmdet3d=0.11.0时  一直出现错误

[image]   请问坐着  这是为什么啊

Ge-huihui, 2024-11-12:
如果不安装mmdet3d   我直接 pip install  - e .   他会报错  如下，想请教一下作者  这是什么原因啊
[image]

donglanxue, 2024-11-12:
closed里面有个和你一样的问题的，你去看一下，cuda11.0和torch1.7.1，mmcv用cuda11.5下的1.2.7

Ge-huihui, 2024-11-12:
[image]
我看了  是这个方法   但是我从源码上一直下载不下来，会显示如下界面，会不会是我的cuda的版本时11.5的  但是下载的pytorch的版本时cuda11.0版本的，还是说我下载cuda的时候没有下载cuDNN的原因啊
[image]
huo'zhe'sh或者是这个情况
[image]

donglanxue, 2024-11-12:
下不了是你源的问题，你换个源就行。cuda11.5不影响，我也试了。直接用原来的setup应该也是可以的，不行你再从源码下载安装

yichen928, 2024-11-27:
> 下不了是你源的问题，你换个源就行。cuda11.5不影响，我也试了。直接用原来的setup应该也是可以的，不行你再从源码下载安装

感谢帮助回复！应该是网络或者源的问题。

closed 2024-11-27 by yichen928